micro-ROS/micro_ros_stm32cubemx_utils #162: Huge library problems

antoniocaso, 2024-12-20:
Hello everyone, and hello @pablogs9 . This time I’m writing here about a problem that’s probably much bigger than the last one, which has been keeping me busy for weeks and has pushed my thesis way beyond the expected timeline. So, a quick initial premise for context: I’m using microROS on my STM Nucleo H723ZG because I need to interface with a vehicle equipped with an onboard computer that already uses ROS2. This vehicle operates on a 20 ms cycle, and so on my MCU I need to configure subscriptions to make (2 of them), which ideally should also occur every 20 ms. I also need to send data to the onboard computer, meaning there will also be a need to publish messages from the MCU to the onboard computer.

Well, my goal would be to achieve real-time timing and guarantees, so as to have publications that occur every 50 ms and reception callbacks that occur every 20 ms. The problem is that using rclc_executor_spin_some() throws everything into disarray, and something even stranger happens: the reception callbacks on the MCU cycle sometimes every 15-16 ms, inexplicably even before the onboard computer (simulated by simply running a publishing node on my personal PC with a timer every 20 ms) can publish! Moreover, as if that weren’t enough, these cycle times aren’t even stable, ranging from 15 ms to 53 ms (sometimes it’s 15, then 16, 17, 18, 19, then 25, 21, 23, and so on), and this doesn’t just happen with the callbacks but also with all the other tasks and cycles, even publication. However, I must point out that when I disable spin_some(), it executes perfectly and precisely (the pubber task) every 15 ms, and NEVER varies, so the problem is definitely inherent to spin_some().

To further describe the problem, it arose when I (foolishly) tried to use TRANSIENT_LOCAL instead of VOLATILE. Before making this change (applied both on the Ubuntu side and on the MCU side, of course), everything was going great: precise, punctual, and with reasonable timing (callbacks were triggered every 25 ms, which is fair because time is needed to execute the code), but after changing, all this madness started, and clearly going back to VOLATILE didn’t solve anything, as if the agent had been permanently corrupted by my reckless move. The reason for this move, however, was that I had noticed that the subscription task was cycling every 2-3 ms, way too fast compared to the 20 ms I had set as the second parameter of spin_some(), and in investigating this problem, I ended up empty-handed.

To try to solve it, I vainly attempted to use spin_period() instead of for(;;) in the subscription task. However, the problem is that when I compile, I get this error:

I tried including unistd.h, but to no avail. I then tried adding this to the code (freertos.c):
#ifdef usleep
#undef usleep
#endif

void usleep(uint32_t useconds) {
    HAL_Delay(useconds / 1000);
}

But it still didn’t work because it overlapped with the usleep definition in unistd.h. At first, it said it couldn’t see the usleep from unistd.h, and now it can, just to mess with me! By removing the unistd.h include, I launched again, but the result is perhaps even more depressing and dramatic than before. Here’s an example, where I count how many times the cycle times exceed a threshold (but mostly I use it to see how much the values fluctuate without even remotely staying stable):
Category: PUBBER
Total numbers found: 62515
1 instances of number 3731.0
62514 instances of number 15.0

Category: SUBBER
Total numbers found: 62515
62515 instances of number 3722.0

Category: Arbiter Callback
Total numbers found: 62515
2 instances of number 3734.0
1 instances of number 100.0
8 instances of number 80.0
3 instances of number 79.0
2 instances of number 77.0
4 instances of number 72.0
2 instances of number 67.0
5 instances of number 62.0
19 instances of number 61.0
5511 instances of number 60.0
218 instances of number 59.0
21 instances of number 58.0
28 instances of number 57.0
51 instances of number 56.0
37 instances of number 55.0
29 instances of number 54.0
36 instances of number 53.0
56 instances of number 52.0
34 instances of number 51.0
35 instances of number 50.0
34 instances of number 49.0
63 instances of number 48.0
41 instances of number 47.0
49 instances of number 46.0
44 instances of number 45.0
57 instances of number 44.0
44 instances of number 43.0
35 instances of number 42.0
310 instances of number 41.0
45496 instances of number 40.0
200 instances of number 39.0
63 instances of number 38.0
52 instances of number 37.0
65 instances of number 36.0
57 instances of number 35.0
68 instances of number 34.0
74 instances of number 33.0
79 instances of number 32.0
55 instances of number 31.0
46 instances of number 30.0
68 instances of number 29.0
72 instances of number 28.0
64 instances of number 27.0
60 instances of number 26.0
60 instances of number 25.0
82 instances of number 24.0
51 instances of number 23.0
64 instances of number 22.0
192 instances of number 21.0
8730 instances of number 20.0
9 instances of number 19.0
26 instances of number 18.0
3 instances of number 17.0

Category: FDCAN Task
Total numbers found: 62515
62515 instances of number 33.0

Category: Info Callback
Total numbers found: 62514
3 instances of number 3742.0
12 instances of number 80.0
6 instances of number 79.0
10 instances of number 62.0
17 instances of number 61.0
6500 instances of number 60.0
230 instances of number 59.0
18 instances of number 58.0
7 instances of number 57.0
3 instances of number 56.0
13 instances of number 55.0
12 instances of number 54.0
23 instances of number 53.0
13 instances of number 52.0
21 instances of number 51.0
11 instances of number 50.0
7 instances of number 49.0
11 instances of number 48.0
15 instances of number 47.0
12 instances of number 46.0
17 instances of number 45.0
16 instances of number 44.0
26 instances of number 43.0
59 instances of number 42.0
671 instances of number 41.0
43564 instances of number 40.0
602 instances of number 39.0
55 instances of number 38.0
56 instances of number 37.0
42 instances of number 36.0
23 instances of number 35.0
22 instances of number 34.0
18 instances of number 33.0
27 instances of number 32.0
37 instances of number 31.0
56 instances of number 30.0
65 instances of number 29.0
35 instances of number 28.0
35 instances of number 27.0
31 instances of number 26.0
22 instances of number 25.0
17 instances of number 24.0
27 instances of number 23.0
52 instances of number 22.0
238 instances of number 21.0
9726 instances of number 20.0
7 instances of number 19.0
20 instances of number 18.0
4 instances of number 17.0

So, in conclusion, all the problems converge: the spin_some() function that doesn’t give me a moment of peace, and perhaps an agent that’s gone crazy.

Please help me, and I apologize for any inaccuracies. Thank you in advance, and if you believe in it, may God bless you Attached at the end, you’ll find the publishing file on the Ubuntu side, the subscription file on the Ubuntu side, and the code running on the MCU (both subscription and publication, all in freertos.c).

[freertos.txt](https://github.com/user-attachments/files/18210501/freertos.txt)
[ubuntu_pubber.txt](https://github.com/user-attachments/files/18210502/ubuntu_pubber.txt)
[ubuntu_subber.txt](https://github.com/user-attachments/files/18210503/ubuntu_subber.txt)

antoniocaso, 2024-12-22:
Edit: I tried to delete and rebuild from scratch the agent, and nothing changed, so I guess it's not a problem of agent-gone-mad.

DrMarkusKrug, 2025-01-15:
Just a short comment. Don't use HAL_Delay() if you are using FreeRTOS with STM32CubeIDE (or CubeMX). They propably share the same timer and therefore locking/influencing each other. This is not a uros problem - more a problem of the HAL layer from STM.

Best Regards
Markus

antoniocaso, 2025-01-16:
> Just a short comment. Don't use HAL_Delay() if you are using FreeRTOS with STM32CubeIDE (or CubeMX). They propably share the same timer and therefore locking/influencing each other. This is not a uros problem - more a problem of the HAL layer from STM. Additionally it is useless to call HAL_Delay() with fractions of milliseconds. Its smallest duration is 1 millisecond. You cannot go below. If you need to be below you have to write you own delay function.
> 
> Best Regards Markus

Yeah my bad, you're absolutely right! Nonetheless, I ditched the whole act of trying to use a different spin() than the spin_some() because no progress was being made, and right now I'm still stuck trying to have this whole plague sorted out.