micro-ROS/micro_ros_stm32cubemx_utils #166: Anyone know how to get past the docker error microcdr

pa0896, 2025-03-03:
Hi I am using this command to get the libmicroros folder in my project-
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy

It runs but I get this error for microcdr which I am unsure about. I saw another thread for the same issue [https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/issues/1274](url) and it says to use a downgraded version but there is no versions here.
I tried changing the branch from jammy to foxy, but it still gives an error.

This is the log output and I am trying to implement microROS for STM32F407VGT6 Discovery Board-

foxy: Pulling from microros/micro_ros_static_library_builder
99803d4b97f3: Pull complete 
db5ff3f6c5dd: Pull complete 
2b302bcd4745: Pull complete 
11384cc19c20: Pull complete 
7c8e1899a5dd: Pull complete 
0ed172a7f8a3: Pull complete 
697c4bfe41b1: Pull complete 
275738f86033: Pull complete 
6c0cfeddf324: Pull complete 
c3049fa8b211: Pull complete 
288a0278b454: Pull complete 
6de3a64d251a: Pull complete 
4f4fb700ef54: Pull complete 
1cf5e6463eb9: Pull complete 
6611c436a5e4: Pull complete 
59286b1a131c: Pull complete 
Digest: sha256:bda0dcca059534cfc39f1fb2744bf58807acb8c6215a33f17fed2a9af15501a7
Status: Downloaded newer image for microros/micro_ros_static_library_builder:foxy
docker.io/microros/micro_ros_static_library_builder:foxy
dos2unix: converting file /project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/library_generation.sh to Unix format...
Found CFLAGS:
-------------
-ffunction-sections -fdata-sections -DSTM32CUBEIDE -DENOTSUP=1 -DECANCELED=1 -DEOWNERDEAD=1 -DENOTRECOVERABLE=1 -mcpu=cortex-m4 -mfloat-abi=hard -mthumb -mfpu=fpv4-sp-d16 -O0 -DUSE_HAL_DRIVER -DSTM32F407xx -DDEBUG --specs=nano.specs -I/project/Middlewares/Third_Party/FreeRTOS/Source/include -I/project/Drivers/STM32F4xx_HAL_Driver/Inc -I"/home/ampellicht/STM32Projects/Game -I/project/Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I/project/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -I/project/Core/Inc -I/project/Drivers/CMSIS/Include -I/project/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -I/project/Drivers/CMSIS/Device/ST/STM32F4xx/Include
-------------
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [128 kB]
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4685 B]
Get:4 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1193 kB]
Get:5 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1301 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal-updates InRelease [128 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal-backports InRelease [128 kB]
Get:8 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:9 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [4252 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:11 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:12 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:13 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [30.9 kB]
Get:14 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [4389 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [4577 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1593 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [34.6 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [4730 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [55.2 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [28.6 kB]
Fetched 35.7 MB in 1min 11s (499 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils-arm-none-eabi libnewlib-arm-none-eabi libnewlib-dev
  libstdc++-arm-none-eabi-newlib
Suggested packages:
  libnewlib-doc
The following NEW packages will be installed:
  binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi
  libnewlib-dev libstdc++-arm-none-eabi-newlib
0 upgraded, 5 newly installed, 0 to remove and 341 not upgraded.
Need to get 271 MB of archives.
After this operation, 2340 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal/universe amd64 binutils-arm-none-eabi amd64 2.34-4ubuntu1+13ubuntu1 [2571 kB]
[... 327 lines cut ...]
/bin/sh: 1: Syntax error: Unterminated quoted string
make[5]: *** [CMakeFiles/microcdr.dir/build.make:118: CMakeFiles/microcdr.dir/src/c/types/sequence.c.obj] Error 2
/bin/sh: 1: Syntax error: Unterminated quoted string
make[5]: *** [CMakeFiles/microcdr.dir/build.make:105: CMakeFiles/microcdr.dir/src/c/types/array.c.obj] Error 2
make[4]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microcdr.dir/all] Error 2
make[3]: *** [Makefile:133: all] Error 2
make[2]: *** [CMakeFiles/ucdr.dir/build.make:115: ucdr-prefix/src/ucdr-stamp/ucdr-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/ucdr.dir/all] Error 2
make: *** [Makefile:87: all] Error 2
---
Failed   <<< microcdr [1.20s, exited with code 2]
Aborted  <<< tracetools_read [0.31s]
Aborted  <<< libyaml_vendor [0.25s]
Aborted  <<< tracetools [1.18s]
Aborted  <<< tracetools_trace [1.17s]
Aborted  <<< rosidl_adapter [1.22s]
Aborted  <<< test_interface_files [1.15s]
Aborted  <<< tinydir_vendor [14.4s]

Summary: 2 packages finished [14.7s]
  1 package failed: microcdr
  7 packages aborted: libyaml_vendor rosidl_adapter test_interface_files tinydir_vendor tracetools tracetools_read tracetools_trace
  1 package had stderr output: microcdr
  53 packages not processed
ampellicht@pop-os:~/STM32Proje

techminion940118, 2025-03-03:
Hello, I am glad to contribute your project and help to resolve this issue.

I think you are facing with libyaml_vendor libraries while pulling docker image. If you are possible to share your docker file, I can assist you to resolve it.

Thank you.

pa0896, 2025-03-03:
hey @techminion940118 thanks for the reply.

i am not well versed with docker so can you tell which file you are asking for? because i am running this command 
docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/ 

i am running it on a terminal in the root of my project folder and as far as I understand, there is no additional file being generated, maybe because of the microcdr error. And as you can see the summary, it says 7 packages aborted, 1 failed and 53 not processed.

techminion940118, 2025-03-03:
Figuring out your command and logs of docker container, this issues is causing from docker run commands.

`docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide`

[image]

You need to validate this command following step.
`echo ${workspace_loc:/${ProjName}}`

And it requires to manipulate this commands with your environment. 

Thank you.

techminion940118, 2025-03-03:
@pa0896

Thanks for clarifying. Can you reach out this file? 

`/project/micro_ros_stm32cubemx_utils/microros_static_library_ide/library_generation/colcon.meta`

You can check this path into Docker container following step.
1. `docker ps`
2. `docker exec -it [YOUR DOCKER CONTAINER NAME] bash`
3. you can navigate this path

Thank you.

techminion940118, 2025-03-03:
Is it working?

pa0896, 2025-03-04:
hey @techminion940118 
i was actually busy with another task so I could'nt apply the steps for docker which you provided. I am actually getting another error which is not the microcdr error which I raised. it is `docker: invalid reference format`, so I am thinking that I am not inputting the path variable correctly, so I think I need to correct that first. I need some time for that since busy with college stuff.

Also, I am not too familiar with docker so I hope that this would not be a stupid question to ask but by using `docker ps`, I get 
the list of all the containers running, but since I have no container running before I run this command `docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/` 

are u saying that i should run `docker exec -it [YOUR DOCKER CONTAINER NAME] bash` while the above command is running, so that i can get the container id using the `docker ps` command and then apply the cd command to reach the colcon.meta file while inside bash?

techminion940118, 2025-03-04:
I think your docker container is not running now.
Can you share your Dockerfile please?

techminion940118, 2025-03-04:
Also it would be helpful for me if you have time to let me know what you are going to do and what purpose of this project.

pa0896, 2025-03-09:
hey @techminion940118 
sorry for the late reply.
so i was having difficulties with the location of my project, since for some reason, the docker pull command could not find the directory which I was pointing it towards. I asked help of some people who were able to fix it for me and I was able to successfully grab the libmicros folder. 
The purpose of the project is to give commands to the servo motors which are connected to the microcontroller whenever a python program provides some input, so basically a python program runs in the background, I press a key on the keyboard, the python program sends a command to microros, microros relays info to the microcontroller, microntroller sends command to servo motor.

closed 2025-03-09 by pa0896