micro-ROS/micro_ros_stm32cubemx_utils #168: ROS 2 Repository Public Key Error During ST Nucleo micro-ROS Setup

Tejasva108, 2025-07-07:
Hello Everyone
While setting up my ST Nucleo board for micro-ROS development on Ubuntu, I am encountering a persistent public key error with the ROS 2 repository. Specifically, when I run the command:

docker pull microros/micro_ros_static_library_builder:foxy && docker run --rm -v ${workspace_loc:/${ProjName}}:/project --env MICROROS_LIBRARY_FOLDER=micro_ros_stm32cubemx_utils/microros_static_library_ide microros/micro_ros_static_library_builder:foxy

(as described in micro-ROS STM32CubeMX utilities), I receive an error stating that the public key F42ED6FBAB17C654 is not available and the ROS 2 repository cannot be verified. Because of this, APT disables updates from the ROS 2 repository, preventing me from installing or updating the necessary ROS 2 packages for my project.

I have tried multiple solutions suggested online, including re-adding the repository, updating the GPG key, and checking for duplicate entries, but none have resolved the issue. I would appreciate clear guidance on how to fix this public key error so I can proceed with my ST Nucleo board setup.

4ntn, 2025-07-08:
Hello @Tejasva108,

This problem is caused by the [ROS signing key expiration](https://discourse.ros.org/t/ros-signing-key-migration-guide/43937). Docker images older than the 1st of June will not work due to this.

Maintained releases (Humble, Jazzy, Kilted, and Rolling) were already updated, so the easy solution is to switch to any of those.

How critical is it for you to continue using Foxy?

Tejasva108, 2025-07-09:
Hi @4ntn 
Thank you for your reply 
I was using a jetson xavier which is on ros2 foxy and I can't update to the newer versions 
I can only use Foxy right now

closed 2025-07-09 by 4ntn