micro-ROS/micro_ros_stm32cubemx_utils #169: Installation Issue related to stm32 ROS2 Foxy

Tejasva108, 2025-07-10:
Hello everyone
I was trying to setup the st nucleo board using the updater docker for ros2 foxy.
However while running the docker I found this issue

[image]
There is a cmake versioni issue but my system is running on 3.26 which meets the requirements
I think this is a issue from the docker side
Is there any way it can be fixed?

4ntn, 2025-07-11:
Hello @Tejasva108,

I understand you are running the `micro_ros_static_library_builder` docker over the `micro_ros_stm32cubemx_utils` files.

Make sure you are cloning the foxy branch from this repository.

If you try to build `micro_ros_stm32cubemx_utils` Kilted with the foxy docker, it will fail as the docker has CMake 3.16.3, which is the version in Ubuntu 20.04, and Kilted requires higher CMake version.

Tejasva108, 2025-07-11:
Hi @4ntn 
Thank you for you reply 
I hadn't changed to the foxy version after git clone and now the docker was installed successfully
However i was facing one final issue while setting up the st nucleo 
While building the error I got this issue

[image]
Is this issue also related to the recent key changes or something else?
Thanks for your help

Tejasva108, 2025-07-16:
Hi @4ntn ,
Sorry for the inconvience caused
I was actually trying to build the firmware for st nucleo board
I am using ST H743ZI
I was trying to run these commands
ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
ros2 run micro_ros_setup create_firmware_ws.sh zephyr olimex-stm32-e407
ros2 run micro_ros_setup create_firmware_ws.sh zephyr ST Nucleo H743ZI
ros2 run micro_ros_setup create_firmware_ws.sh freertos ST Nucleo H743ZI
But i was getting this error

 ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407 --rosdistro=foxy
Creating firmware for freertos platform olimex-stm32-e407
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
Cloning into '.'...
=== ./ros2/test_interface_files (git) ===
Cloning into '.'...
=== ./ros2/unique_identifier_msgs (git) ===
Cloning into '.'...
...........
=== ./eProsima/Micro-CDR (git) ===
Cloning into '.'...
=== ./eProsima/Micro-XRCE-DDS-Client (git) ===
Cloning into '.'...
=== ./ros2/tinydir_vendor (git) ===
Cloning into '.'...
=== ./uros/micro_ros_msgs (git) ===
Cloning into '.'...
=== ./uros/rcl (git) ===
Cloning into '.'...
[... 137 lines cut ...]
Downloading ARM compiler, this may take a while
=== ./freertos_apps (git) ===
Cloning into '.'...
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rclc_parameter: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rcl: Cannot locate rosdep definition for [rcpputils]
rclc: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rmw: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rosidl_typesupport_microxrcedds_c_tests: Cannot locate rosdep definition for [rosidl_typesupport_introspection_c]
libyaml_vendor: Cannot locate rosdep definition for [rcpputils]
rcutils: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rcl_lifecycle: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
tracetools_test: Cannot locate rosdep definition for [launch_ros]
ros2trace: Cannot locate rosdep definition for [ros2cli]
rmw_implementation: Cannot locate rosdep definition for [rcpputils]
rosidl_typesupport_c: Cannot locate rosdep definition for [mimick_vendor]
rosidl_default_generators: Cannot locate rosdep definition for [rosidl_typesupport_introspection_cpp]
rclc_lifecycle: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rosidl_default_runtime: Cannot locate rosdep definition for [rosidl_typesupport_introspection_cpp]
test_rmw_implementation: Cannot locate rosdep definition for [rmw_dds_common]
rcl_logging_noop: Cannot locate rosdep definition for [launch_testing]
tracetools_launch: Cannot locate rosdep definition for [launch_ros]
rcl_action: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
rosidl_typesupport_cpp: Cannot locate rosdep definition for [rcpputils]